gargomoma/fakeTec_pcb #2: Add a drill hole below where the boost pads sits

Nestpebble, 2024-10-04:
Hey, could you add a little hole underneath the boost pads, please? It's sort of shown in this drawing:
https://discord.com/channels/867578229534359593/1194757507013427250/1272294175194222593

I've also made a footprint in the EasyEDA library that shows it - I'll find a link a bit later on.

lupusworax, 2024-11-05:
I would also love to see  that happen, please  please please! It would make things a lot easier.

lupusworax, 2024-12-06:
Amazing! Thanks for updating the design, the reset an function button is a perfect upgrade also the cutout, I ordered right away.  With the updated HT-RA62 modules now finally working reliable the faketec is going to be total success story for sure!

Nestpebble, 2024-12-06:
Yep - all looks good to me!

closed 2024-12-06 by Nestpebble