gargomoma/fakeTec_pcb #5: Dummy model

candre23, 2024-12-05:
I'm in the process of designing a case for this board, and I ended up creating a full dummy model STL to help with that.  It's PD, so feel free to include it in this repo if you like. 

https://www.printables.com/model/1098558-faketec-dummy-model

lupusworax, 2024-12-12:
Very nice thanks for the model!  You can make it even more low profile if you simply stick the pin array in from the other side leaving out the black spacer bar in between boards. With the new version 3 of the PCB board with the booster pins accessible, there is no need for the gap anymore ;)

candre23, 2024-12-12:
I considered that, but my eyes are too old for SMD soldering, and the through-hole resistors (and battery connector wires) had to go somewhere.  With the pin headers raising the promicro up a bit, they can go on that side between the promicro and the PCB.  If they were on the back, then you'd need to account for that in the case design.

While we're mentioning it, it sure would be nice if there were dedicated battery through-holes somewhere more convenient on the PCB.  Having to just sort of bodge them onto the SMD pads feels sloppy.

candre23, 2024-12-13:
And in case you're interested, this is the enclosure I made using the dummy: https://www.printables.com/model/1099391-p4gr-a-minimalist-enclosure-for-the-faketec-meshta

lupusworax, 2024-12-13:
@gargomoma  should totally add your design as well to the Details ;)