gargomoma/fakeTec_pcb #6: FYI Oscillators/layout have/has been swapped/changed with better ones on HT RA62 They do support 200byte messages now!!

lupusworax, 2024-12-06:
Oscillators and even the layout has been changed, they actually do work now! 
They also have a new design print on the back of the PCB  compared to the old modules.

lupusworax, 2024-12-06:
I am still testing but it is looking very good as of now!  Gonna make several more boards and will report back.

lupusworax, 2024-12-06:
[image]
[image]
[image]

gargomoma, 2024-12-08:
Awesome! When did you bought the latest modules?
Do you see any physical difference between the old ones? It would be great to spot the "good" ones before installing.

lupusworax, 2024-12-08:
I bought them 2 ish weeks ago I made a post on the meshtastic reddit about it showing the visible differences  also made a video to prove it. https://www.reddit.com/r/meshtastic/comments/1h9gq2b/new_htra62_sx1262_lora_modules_spotted_in_the/

JohnTaipei, 2024-12-28:
good job
i have older version