gargomoma/fakeTec_pcb #8: My enclosure designs made for the Faketec

lupusworax, 2024-12-12:
I just wanna link some of my models I made lately that were made for the Faketec. 

The "FAKECAP": 
https://www.printables.com/model/1067553-meshtastic-fakecap-super-slim-nrf52-gps-node

The "Tamameshi": 
https://www.printables.com/model/1061194-lwc-meshtastic-tamameshi-portable-faketec-node

The " Meshformer":
https://www.printables.com/model/1058668-lwc-meshtastic-meshformer-communicator-node

The " Waypoint": COMMING SOON

gargomoma, 2024-12-12:
I'll add it to the readme thanks!!

lupusworax, 2024-12-12:
No, THANK YOU! the faketec is freaking amazing and you are awesome ;)

lupusworax, 2024-12-12:
The "Waypoint" : https://www.printables.com/model/1104742-meshtastic-waypoint-non-permanent-outdoor-solar-as

lupusworax, 2024-12-21:
And one more for Team Faketec ;)
https://www.printables.com/model/1116459-meshtastic-meshtak-tacticool-big-boi-water-resista

lupusworax, 2024-12-29:
And one more design of mine: 
https://www.printables.com/model/1123408-meshtastic-faketec-f1-micro-node-nrf52

lupusworax, 2025-01-11:
And another one: 
https://www.printables.com/model/1144370-meshtastic-faketak-scout-ultra-portable-tacticool

lupusworax, 2025-02-01:
Update on my Fakecap Node with a "special edition" with Werwolf tribal imprint and vibra motor update: 

[image]

[image]

lupusworax, 2025-02-01:
Are you powering the gps module permanently or with a mosfet/ transistor? If you use a transistor/mosfet which one exaclty is it?  To take stress off the 3.3 volt rail I connect my GPS + directly  to RAW as the module can take up to 5 volts. (it still is switched off with a mosfet disconnecting it from ground)  Does the fake Tec recognize the gps module but it keeps searching for a fix? 
I also encountered an issue with the fake tec that if you wrongly connected  TX and RX it does not help to change the pins in the app, I always have to physically swap RX/ TX either on the NRF dev board or on the gps module.

lupusworax, 2025-02-01:
> After fixing it should at least blink its led. I didn't mess anything with wiring, I used this module standalone. I thought it's 3,3V only, even confirmed it on the internet. Are you sure it is 5V tolerant? Anyway, with my faketek's 3,3V LDO it won't work. Maybe it's just doa. Thanks!

RAW does not forward the USBs 5 volts but only the battery directly ( so about 4.2 volts max) I learned this the hard way when I was also pulling my hair as GPS would not power on over USB  with no battery attached XD

oldmanav, 2026-01-21:
Daft question, apologies if this is listed somewhere and I've not spotted it. How did you print the Tamameshi case? Did you use supports and in what orientation please?

lupusworax, 2026-01-22:
Sorry read your question just now, yep standing upright, your 100% correct.