gargomoma/fakeTec_pcb #9: boost pads hole, off center

lupusworax, 2024-12-17:
I either have a different pcb layout or the hole is not correctly aligned with the boost pad passthrough. 
[image]

gargomoma, 2024-12-17:
Hi!
Yes, seems the gerber was saved with the wrong dimensions.
I didn't saw this since my boards are sent by the slow shipment option.

 Sorry T_T

I just fixed the hole, I'll upload the _good_ one during the day.

[image]

lupusworax, 2024-12-17:
no worries just like before I am using a 0,1mm thin wire to bridge it ;)
[image]