gargomoma/fakeTec_pcb #12: Specific values for R1 and R2 resisors

OTLabs, 2024-12-29:
Hi!

I see that you use 1M resistors for R1 and R2.

Should both of them have the same value?

Could you please comment about possible range of values they may have?

Thank you for so great board!

lupusworax, 2024-12-29:
I am using 680K  and 1M, and then its playing around with the power calculator on the Meshtastic docu: https://meshtastic.org/docs/configuration/radio/power/

OTLabs, 2024-12-29:
@lupusworax Thank you for clarifying! Let me confirm: R1 is 680K and R2 is 1M, right?

lupusworax, 2024-12-29:
In my case yes but the point is it does not have to be those values,  the only reason using a voltage divider on the faketec is that the max voltage the NRF52 can handle is 3.3 volt, so a full lipo battery with 4.2 or above 3.3 volt in general would damage the chip, thats the only reason a voltage divider is used in this case. 

If it was an Arduino  uno for example thats perfectly fine with up to 5 volts on any gpio you can directly connect a 1s lipo to a gpio and measure the voltage there. 

 The point is to keep the voltage withing a save value and as long you know both exact values of those resistors you can use math to determine the actual voltage of the battery in relation to the voltage drop between the resistors.  You can use any other Resistors as long the voltage  between those two does not surpass 3,3 volts.  with the ADC multiplier you then can calibrate the used battery in relation to the used voltage divider (resistor values).  That is also the reason why the battery musted be charged up to 100% ( 4,2 volts)  when you do the ADC multiplier calculation I posted before.

OTLabs, 2024-12-29:
Thank you for explanation! I have not find any schematics so I have my doubts about connections going to and from this divider, looks like it follows classic R1/R2 divider approach.

So 680k/1M divider will give me 2.5V output for 4.2V input. Is it ok? I mean, should I get closer to 3.3V? Will I benefit, somehow, from getting 3.0V (1.2M/3M), 3.1V (2.2M/6.2M), or 3.2V (300k/1.1M) output?

I used [Texas Instruments Voltage Divider Calcularor](https://www.ti.com/download/kbase/volt/volt_div3.htm) to get these numbers.

lupusworax, 2024-12-29:
like I said it does not matter, the ADC multiplier value always  needs to be set when the Battery is at max ( 4.2 volts) so it does really matter if the divider output is 2,5 or 3 or 3.2 as long it knows that 2.5volt on the divider or any other result of the voltage devider  equals 4.2 volts in relation to the ADC multiplier in the power settings.  have you found the  calculator in the link I sent you? 
[image]

lupusworax, 2024-12-29:
I forgot, when you power on your Taketec node ( whatever resistors you used for the divider) and it does not show any voltage at all, way to much or only that you are "only" connected to external power, you have to input a ADC multiplier in the app on your phone to get pretty much any reading that is not above 4.2 volts.   best would be somewhere in the middle.

 Depending if your battery is full or empty you have to charge it until it stops ( blue led goes off) unplug it then and give it a bit until it settles on a voltage / charge. It has to be below 4.2 volts!!!  ( or perfectly spot one but I never managed to do this by accident xD )  once its stable and does not change anymore you use the calculator above and once you enter the new calculated Adc multiplier value  it should be pretty much spot on.

OTLabs, 2024-12-30:
I see the point, so I can use practically any divider which gives me 3.3V max output, and calibrate at fully charged battery to 100% with the calculator. Yes, I got the link to the calculator and already calibrated my T-Beam node. Thanks a lot!

lupusworax, 2024-12-30:
> I see the point, so I can use practically any divider which gives me 3.3V max output, and calibrate at fully charged battery to 100% with the calculator. Yes, I got the link to the calculator and already calibrated my T-Beam node. Thanks a lot!

Spot on! Could not have summarized it better myself ;)  You r more then welcome!

dtjager, 2025-08-03:
> Spot on! Could not have summarized it better myself ;) You r more then welcome!

What do I do if when I plug the battery in and connect and it just says "powered"?

xia0, 2025-10-05:
> What do I do if when I plug the battery in and connect and it just says "powered"?

> If "Battery Charge Percent" (e.g., B 3.82V 60%) is not displayed on the screen, it means that the default value of "Operative Adc Multiplier" is too high. Lower the "Operative Adc Multiplier" to a smaller number (it is recommended to decrease by 0.1) until the screen displays "Battery Charge Percent". Enter the current "Operative Adc Multiplier" in use into the "Operative Adc Multiplier" field in the calculator. Also, input the "Battery Charge Percent" displayed on the screen into the calculator.