gargomoma/fakeTec_pcb #13: NO GPS BUG

jack-changyan, 2025-01-05:
Hi bro, thanks a lot for bringing this project to us, it's really cool. But recently I've encountered a problem. I'm using the latest nice_nano_bootloader-0.9.2_s140_6.1.1.zip and this version of the firmware nrf52_promicro_diy_xtal-v2.5.19.6aabbedc.daily.uf2, and I'm getting a "NO GPS" issue. I've tried swapping the TX and RX of the GPS module several times, but it still doesn't work properly. Have you guys encountered this problem?

gargomoma, 2025-01-06:
#7 
Didn't updated yet to last version, nor use gps on mine.
Could it be a bug on their code?

lupusworax, 2025-01-07:
did you swap GPS RX/TX pins only in the app or did you also swapped them on the board directly?  If not give it a try.

lupusworax, 2025-02-08:
It could help others if you told everyone what the issue was. Nothing wrong with asking for help but once you get it, it would be nice to tell everyone who tried to help what the problem was instead of just closing the issue.

ChangYanChu, 2025-02-25:
@lupusworax  I agree with your point. I updated to the nrf52_promicro_diy_xtal-v2.5.19.f1a89002.daily.uf2 version at that time, and simultaneously connected GPS RX to PCB RX and GPS TX to PCB TX. Then the GPS module could work normally. It's really amazing. In my past understanding, RX and TX were always connected in a crossed manner.

JudithWang25, 2026-02-04:
I also had a GPS problem today. I once suspected I had bought the wrong GPS antenna, but thankfully I saw this article.

I am using a V5 verB circuit board and an ATGM336H GPS module.

The GPS pins are RX→RX, TX→TX. Then, following the original author's settings, I changed RX→20, TX→22, EN→24, but it's not working.

It only worked successfully when I changed RX to 22 and TX to 20.

Although @ChangYanChu has already solved the problem, I hope my experience can help others...

[image]

[image]

[image]

[image]

lupusworax, 2026-02-04:
I guess you are using a firmware  2.7.16/ 2.7.17/2.7.18? thats where they changed the pin configuration, also I cant get my navigation switch to work at all with any of the listed firmware versions above, no clue what they fucked up but  its meshtastic so Its expected that everything gets fucked overboard with newer firmware releases.

lupusworax, 2026-02-04:
> > I guess you are using a firmware  2.7.16/ 2.7.17/2.7.18? thats where they changed the pin configuration, also I cant get my navigation switch to work at all with any of the listed firmware versions above, no clue what they fucked up but  its meshtastic so Its expected that everything gets fucked overboard with newer firmware releases.
> 
> Yes, my firmware version is 2.7.17. I didn't notice the changes made in these versions. Perhaps it's just a bug; we'll wait for the official fix.

I think it was intentionally, at least when it comes to the RX and TX pins, as you were not able to change the pins in software for a long time, you had to physically swap wires to make gps work. swapping ports 20 and 22 in the app did nothing. I think thats what they fixed, so now you can actually swap connections in software successfully but now you can not connect RX to RX  and TX to TX anymore as the board was designed that way  to make it easier for users not knowing that "normally" those  lines need to be crossed  out ( TX on RX , RX on TX).