gargomoma/fakeTec_pcb #24: fakeTec V5 - Battery protection, low profile, JST connector, dedicated battery pins

ShimonHoranek, 2025-03-19:
New updated design adds XB8089D0 for battery protection, JST connector, same as on Heltec V3, dedicated battery pins, improved voltage measurement. 

I have also decided to change the way you mount the Nice!Nano. Now there are SMD pads for low profile and (in my opinion) easier mounting. One way is to solder through Nice!Nano holes for the solder to reach and properly heat the SMD pads. In my testing this works very well. Other option is to sand down the Nice!Nano holes to create castellated pads (sanding fiberglass is dangerous, wear proper safety equipment). 

The design is heavily based on [lupusworax](https://github.com/lupusworax)'s design. Other than the BMS I have also added dedicated pads for the GPS module. (Note that TX on board connects to TX on GPS and RX -> RX).

The XB8089D0 is capable of overcharging, overdischarging, overcurrent, load short circuiting protection etc. This mainly means battery voltage can't get under 2.90 V. This improves stability and prevents memory corruption. Thanks [Dude5101](https://github.com/Dude5101) for help.

There is an jumper on the board. It is important that you short one of the connections. The BMS one connects the BMS, direct connects the battery directly to Nice!Nano bypassing the BMS Circuit. 

And the best for last, if you want to improve this board, I have recreated the board in KiCad (Free EDA program). This means you can now download the design files and add/remove/change whatever you want. The files are licensed under GPLv3. I have tried to make the board as self explanatory as possible. 

Parts list:
- 1x Nice!Nano v2 board or equivalent (same as all fakeTecs)
- 1x Lora radio (same as all fakeTecs)
- 1x XB8089D0
- 3x SI2312
- 2x 3x6x2.5mm SMD buttons
- 3x 1206 10k resistors
- 2x 1206 10M resistors
- 2x 1206 100nF capacitors
- 1x JST 1,25 connector (optional)
- 1x fakeTec v5 board

I'm still waiting for the board to arrive so I have not yet tested it's functionality. 

[fakeTecv5_gerber_files.zip](https://github.com/user-attachments/files/19340467/fakeTecv5_gerber_files.zip)

[fakeTecv5.zip](https://github.com/user-attachments/files/19340476/fakeTecv5.zip)

shurillu, 2025-03-19:
Hello ShimonHoranek,

   very nice job, thank you for sharing!
Just few points: 

- without the hole for accessing the charge boost pads, you can't put the Nice!Nano directly over the PCB because the solder joint on the boost pads doens't have his room on the PCB.
- the buttons should have a pullup resistor. They are not mandatory but is better and safer. Take a look from what [Heltec do for the T114](https://resource.heltec.cn/download/Mesh_Node_T114/MeshNode-T114_V2.0_SCH(2).pdf) and what [LilyGO do for the T-Echo](https://github.com/Xinyuan-LilyGO/T-Echo/blob/main/T-Echo_Schematic.pdf)

Hoping that helps, 

Stefano

ShimonHoranek, 2025-03-19:
Hello Stefano,
Thanks for the suggestions!

- I have added the hole. Good point that now even the slightest solder joint may prevent a good connection of the boards.
- I thought about them when designing, but the previous designs didn't have pull-up either, so I assumed it used internal pull-up. I have added them now along with filtering caps, as shown in the Heltec shematic (Thanks for them). 

I have updated the original post with the updated files and information.

shurillu, 2025-03-27:
Hello Simon, 
   what a nice work! Thank you for sharing it!
I have a doubt regarding the "DIO2 TXEN2" silkscreen: I guess it should be "DIO2" only.

Would you like to add some other functionalities? Here some ideas:

in the opposite side where sit the I2C OLED Display connector,
1. GPS header connector: something like 3V3 - T1 GND - GPS TX - GPS RX. T1 is the drain of the T1 MOSFET. Maybe could be useful a jumper selector (something like you did for the MBS selector) in order to select if the GPS can be powered by T1 MOSFET or always powered (T1 GND directly connected to GND);
2. Buzzer header connector: something like 3V3 - T2. T2 is the drain of the T2 MOSFET;
3. Vibration motor header connector: something like 3V3 - T3. T3 is the drain of the T3 MOSFET;

All that function are mainly for a portable node.
To add more flexibility, wholud you add a solar charger circuit?

4. Solar charger circuit and connector: scaling passive component from 1206 to 0804 package maybe we have enough room to place the solar circuit to power (and recharge) the battery. With this function, this board will be the definitive board. The solution for the solar panel connector could be the same for the battery (an JST 1.25 connector plus a 2.54 header connector). I have some ideas regarding how to implement that, let me know if you're interested;
5. Lastly, the nRF52840 pro micro have 3 unused pin, they could be used to implement canned message functionality (with a rotary encoder, for example). The pins are the ones near the ceramic antenna. As the previous point, I have some ideas, let me know if you're interested.

Again, thank you for your work!

Stefano

hayschan, 2025-03-31:
Excellent work.

Members of our Chinese community - [MeshCN](meshcn.net) has manufactuered the board. It works perfectly.

violarulan, 2025-03-31:
Not sure if it's my soldering skill issue or something. I ordered it and gets Critical error 3 which is LoRa module error.

I measured 2.7v at RA62's VCC and 3.3v at RA62's RST as well as D10. Will try to order some other board and test again. I'll update if things go well.

ShimonHoranek, 2025-03-31:
The boards arrived today. After hand-soldering everything I can confirm everything works. I haven't tested the screen but there should (hopefully) be no reason for it not to work. I will in coming days update the original post with some pictures after I get it to a nice-looking state. Also i will update the design slightly, replacing the GPS EN pin with a VCC pin. I didn't quite understand how the GPS worked when designing and the EN pin is not needed to be exposed. For now you can connect the VCC pin of the GPS module to any 3,3 - 5 V pin (I have it on RAW)

I might also flip the battery JST connector so that the wires face the USB. It seems more practical as the wires won't be sticking from the board.

Regarding @shurillu's suggestions: The DIO2 TXEN silkscreen was on the fakeTec V4 like this and they are connected on the board. It is probably unnecessary  and will be removed with the next update. I will add the jumper and T1 pin next to the gps pins and the 3,3V pins. 
Solar is a bit trickier. I will see what I can do. Definitely like the idea of a complete board that the user can populate with parts based on their usage. 
Encoder for canned messages is definitely possible, will see if i can fit three more pads on the board somewhere.

Dude5101, 2025-04-01:
Very nice well done!

V6 idea add pins for the GPS

[image]

Dude5101, 2025-04-01:
> Very nice well done!
> 
> V6 idea add pins for the GPS Can then be used with ATGM336H
> 
pins should be mirrored so that ATGM336H antenna connect is accessible

ShimonHoranek, 2025-04-01:
I'm currently using it with ATGM336H. There are exposed circular pads on the backside of the board (The VCC pin is currently missing, will add and change it a bit, for now use some 3,3 - 5 V pin on the board). This means you will need to solder some wires between the boards. The problem with TH pins is that they take up a lot of space and occupy both layers. I have played with adding more TH pins but ended up only putting there the display and battery to make space for other components and better routing.

shurillu, 2025-04-02:
> The boards arrived today. After hand-soldering everything I can confirm everything works. I haven't tested the screen but there should (hopefully) be no reason for it not to work. I will in coming days update the original post with some pictures after I get it to a nice-looking state. Also i will update the design slightly, replacing the GPS EN pin with a VCC pin. I didn't quite understand how the GPS worked when designing and the EN pin is not needed to be exposed. For now you can connect the VCC pin of the GPS module to any 3,3 V pin

I agree with you: the GPS EN should be the signal that drive the T1 MOSFET for turning on/off the GPS module and, for this reason, it should not be exposed. This functionality is used mostly for battery powered nodes (to save battery).

> I might also flip the battery JST connector so that the wires face the USB. It seems more practical as the wires won't be sticking from the board.

Great idea, it helps to fit the board inside small cases.

> Regarding [@shurillu](https://github.com/shurillu)'s suggestions: The DIO2 TXEN silkscreen was on the fakeTec V4 like this and they are connected on the board. It is probably unnecessary and will be [...]

About solar charger, an MPTT chip could be a best choice, here an [example:](https://github.com/ponzano/MeshLink-hardware/blob/main/Schematic_MeshLink_2025-02-23.pdf) (jump to the Solar-Lipo sheet). 

> Encoder for canned messages is definitely possible, will see if i can fit three more pads on the board somewhere.

Regarding the pads for the rotary encoder, it is possible to have a dedicated GND pad near the other three pad (A, B, button)?
Pullup resistors could be a plus.

Buzzer and vibrator motor (T2 and T3 pads) could have a dedicated 3V3 pad? In order to have easier cabling/soldering.

Again, thanks for all your efforts!

Stefano

ShimonHoranek, 2025-04-02:
> I believe one 1k resistor is missing in the material list above. Can you add it please? [@ShimonHoranek](https://github.com/ShimonHoranek)
You are right, added. And congrats on the fixed board, if I can give you a soldering tip, your joints look like they could use a bit more flux, you will have much better time soldering.

candre23, 2025-04-02:
Would it be at all possible to replace the XB8089D0 with something like a TI BQ25186?  Because the XB8089D0 doesn't really offer that much beyond what the nicenano already does, while the TI chip would provide the option to use LiFePO4 cells in addition to standard lipo.

For those unaware, LiFePO4 is a more stable battery chemistry that is substantially preferable for devices which see wide temperature variances - such as permanent outdoor nodes.  It handles very high and very low temps much better than prismatic Li-ion cells or LiPo pouches.

shurillu, 2025-04-02:
> Would it be at all possible to replace the XB8089D0 with something like a TI BQ25186? Because the XB8089D0 doesn't really offer that much beyond what the nicenano already does, while the TI chip [...]

[ShimonHoranek](https://github.com/ShimonHoranek) may confirm it: the XB8089D0 is a battery protection chip, not a lipo charger chip (that is present on the nrf52840 pro micro board). 
The XB8089D0 is needed because the promicro charger chip (LTH7R) nor the TI BQ25186 have a low voltage battery protection threshold.

Is it correct?

ShimonHoranek, 2025-04-02:
Yes, the XB is there for battery protection (battery reverse, temperature protection, overcharge, overcurrent, overdischarge etc.). This prevents the flash curruption which causes meshtastic node name to reset.

candre23, 2025-04-02:
These features are present on the TI chip as well.

> • Integrated fault protection
> – Input overvoltage protection (VIN_OVP)
> – Battery short protection (BATSC)
> – Battery overcurrent protection (BATOCP)
> – Input current limit protection (ILIM)
> – Thermal regulation (TREG) and thermal shutdown (TSHUT)
> – Battery thermal fault protection (TS)
> – Watchdog and safety timer fault

But I'm not pushing for that TI chip specifically.  I'm more suggesting that if you're going to go to the effort to include an external battery management chip at all, then ideally you should choose a chip that includes the capacity for managing LiFePO4 cells.

Dude5101, 2025-04-02:
> These features are present on the TI chip as well.
> 
> > • Integrated fault protection
> > – Input overvoltage protection (VIN_OVP)
> > – Battery short protection (BATSC)
> > – Battery overcurrent protection (BATOCP)
> > – Input current limit protection (ILIM)
> > – Thermal regulation (TREG) and thermal shutdown (TSHUT)
> > – Battery thermal fault protection (TS)
> > – Watchdog and safety timer fault
> 
> But I'm not pushing for that TI chip specifically. The BQ25186 was merely the first part that popped up in a search for full-featured lipo/lifepo4 BMS chips which were relatively inexpensive. I'm more suggesting that if you're going to go to the effort to include an external battery management chip at all, then ideally you should choose a chip that includes the capacity for managing LiFePO4 cells.

TI chip has no low voltage protection which is the point of using the XB8089. Specifically shutting down below 3V so that the node stays stable

lupusworax, 2025-04-03:
Great work man! love all the nice lettering on the board and added battery safety. This little pcb is becoming crowded! haha Ordered pcb s right away, very excited to build them! Cheers

violarulan, 2025-04-06:
[image]

Some SMD buttons can't fit well onto the pads. In my case the 3\*6 (above) are a little bigger. The 3\*4 (buttom) ones are just fine. Should take a notice on the size while choosing the buttons.

pooch-1, 2025-04-07:
Can anyone provide a link to 1206 100nF 10V capacitors on AE? I can't find anything that matches.

lupusworax, 2025-04-07:
> Can anyone provide a link to 1206 100nF 10V capacitors on AE? I can't find anything that matches.

They dont have to be specific for 10 volts,  as long capacity and size fits. as long they are above 5 volts you should be ok.

pooch-1, 2025-04-07:
> > Can anyone provide a link to 1206 100nF 10V capacitors on AE? I can't find anything that matches.
> 
> They dont have to be specific for 10 volts, as long capacity and size fits, and they are above 5 volts your ok.
> 
> https://de.aliexpress.com/item/1005006870220937.html

Thank you, I'm very new to this.

lupusworax, 2025-04-07:
> > > Can anyone provide a link to 1206 100nF 10V capacitors on AE? I can't find anything that matches.
> > 
> > 
> > They dont have to be specific for 10 volts, as long capacity and size fits, and they are above 5 volts your ok.
> > https://de.aliexpress.com/item/1005006870220937.html
> 
> Thank you, I'm very new to this.

No worries, most importantly with caps is that you DONT use some with lower voltage specs,  then the main voltage  you are using in your build as that would damage them.

meekowai, 2025-04-07:
Please be careful when sanding the nrf board to make castelated holes. There are some copper paths right next to where you want to sand - between the holes.

lupusworax, 2025-04-07:
> Please be careful when sanding the nrf board to make castelated holes. There are some copper paths right next to where you want to sand - between the holes.

I asked the seller  on ali ( tenstar robot store) on several occasions / orders  if they could also offer a SMD variant of the NRF dev board without the THT  but they sent me links for random SMD boards, they simply dont understand what I was trying to say. 

[image]

meekowai, 2025-04-07:
> 5. Lastly, the nRF52840 pro micro have 3 unused pin, they could be used to implement canned message functionality (with a rotary encoder, for example). The pins are the ones near the ceramic [...]

Encoder pins need resistors and caps?

EA7CH, 2025-04-07:
Exists some BOM archive for this board?

meekowai, 2025-04-08:
3-way nav switch is just 3 buttons in one enclosure, right? Why not use same resistors and caps you added to BTN and RST? Why bother with BTN and RST if we wont bother with others?

CLAiMORE, 2025-04-11:
> Found out there is another over-discharge protection IC called FS312F-G

Just on a quick note, I have used this IC in the past with great results (in combination with the old mighty TP4056 IC). Actually I remember that I was deciding between using a dw01a and the FS312F-G and did opt for the latter, but can't remember why xD
Here is the layout I've used, it's working flawlessly in my case. In that project the circuit has to power 16 RGB LEDs (5050WS2812B) aswell as an ATTINY85.
Maybe it helps.

[image]

Dude5101, 2025-04-11:
> FS312F-G

2V low voltage cutoff, how is this a suitable replacement for the XB8089, It has no low voltage protection, which is the point of using the IC in the first place. Even paired with TP4056 you have no brownout protection.

meekowai, 2025-04-11:
> > FS312F-G
> 
> 2V low voltage cutoff, how is this a suitable replacement for the XB8089, It has no low voltage protection, which is the point of using the IC in the first place. Even paired with TP4056 you have no brownout protection.

Are we looking at the same chip from Fortune?
http://ic-fortune.com/upload/Download/FS312F-G-DS-10_EN.pdf

Overdischarge at 2.9v, enable at 3v. In-place replacement for dw01a.

Dude5101, 2025-04-11:
> Are we looking at the same chip from Fortune? http://ic-fortune.com/upload/Download/FS312F-G-DS-10_EN.pdf
>
> Overdischarge at 2.9v, enable at 3v. In-place replacement for dw01a.

Ah! No we were not looking at the same chip.

Nice that is a good option 👍

lupusworax, 2025-04-12:
We ve come a long way since Version1, all variants so far.....even the one with the misaligned boost pad hole xD 

[image]

EA5JQP, 2025-04-13:
Hello everyone,

I have found XB8089D only. Is the footprint the same?

Thank you!

ShimonHoranek, 2025-04-13:
@EA5JQP it may have the same footprint, but it is not a good fit for this board. It's cutoff voltage is too low, which means your board's memory will get corrupted

deanZZZZZ, 2025-04-15:
I can't find the model XB8089D0
XB8089D have 2.5 cutoff voltage
which one should I buy?

[image]

ShimonHoranek, 2025-04-15:
Oh, it appears i too bought a wrong one - XB8789D0. Comparing the datasheets, they both have 2,9 V cutoff. The only significant difference i could find is that the 8789 has overcurrent protection of 14 A and a bit higher power consumption (9 uA), whilst the 8089 has overcurrent protection of 10 A and lower power consumption (6 uA).

pooch-1, 2025-04-15:
Here is a listing I found for XB8089D0.

Disclaimer: Please note I have not bought or used this item....yet

https://a.aliexpress.com/_mL1Ckp9

lupusworax, 2025-04-17:
> Here is a listing I found for XB8089D0. Scroll and select the correct item as the link will not take you to the correct selection.
> 
> Disclaimer: Please note I have not bought or used this item....yet
> 
> https://a.aliexpress.com/_mL1Ckp9

Cant open the affiliate link you posted, do you also have a normal link?

pooch-1, 2025-04-18:
> Cant open the affiliate link you posted, do you also have a normal link?

https://www.aliexpress.com/item/1005006961677070.html?spm=a2g0n.productlist.0.0.1dd9489dk4TKpO&browser_id=b5681ad2464b484da5b60ba456dfb2a1&aff_trace_key=93f3cdf4f70146b59fb1af806d45dfc4-1744938877442-01134-UneMJZVf&aff_platform=msite&m_page_id=fzwiicgqrcazuwud196467672dcec479e2e04b4c55&gclid=&pdp_ext_f=%7B%22order%22%3A%226%22%2C%22eval%22%3A%221%22%7D&pdp_npi=4%40dis%21CAD%2113.98%2113.27%21%21%2172.03%2168.37%21%402101ec1a17449388898173866e7e93%2112000038867578512%21sea%21CA%210%21ABX&algo_pvid=0e1cfeb7-7a3c-4129-bdaa-ea2f597a0342

Does that work?

lupusworax, 2025-04-18:
oh, 404 not available in your country :/ nevermind I will find an alternative, but thanks!

pooch-1, 2025-04-18:
> oh, 404 not available in your country :/ nevermind I will find an alternative, but thanks!

Strange. Here is the item listing and store name. 

[image]

[image]

lupusworax, 2025-04-18:
hmm no clue honestlyI get the follolwing on both links: 

[image]
but dont mind I found an alternative already ;)

EA5JQP, 2025-04-18:
It is because the seller doesn't want to send to your location. Sometimes you can trick AliExpress adding the item from the website but changing the region before

-------- Mensaje original --------
El 18/4/25 17:28, Loup  ha escrito:

ygriega17, 2025-04-22:
https://s.click.aliexpress.com/e/_ooMI2Pz

**Shopping in the Aliexpress coin section**

[image]

lupusworax, 2025-04-22:
Yeah great,  another affiliate link...... would you guys mind stop doing this and post normal links without trying to make money from it?  This is an open source project after all.

pooch-1, 2025-04-22:
> Yeah great, another affiliate link...... would you guys mind stop doing this and post normal links without trying to make money from it? This is an open source project after all.

Can you tell me how to create a link without it turning in to an affiliate link? I am not doing that deliberately and new to AliExpress. When using the app I go to share an item listing and copy the address it provides. I wasn't aware it is creating an affiliate link but I can't find another way to share.

lupusworax, 2025-04-22:
> > Yeah great, another affiliate link...... would you guys mind stop doing this and post normal links without trying to make money from it? This is an open source project after all.
> 
> Can you tell me how to create a link without it turning in to an affiliate link? Do I have to stay in a Web browser? I am not doing that deliberately and new to AliExpress. When using the app I go to share an item listing and copy the address it provides. I wasn't aware it is creating an affiliate link.

"Normally"  ali express links look like this: https://de.aliexpress.com/item/1005006501271520
I Dont use the app, so no clue what was set up in the app.

lupusworax, 2025-04-22:
> I believe that is only the shortened link. By default it looks like that when you want to share a link from the app.
> […](#)

Oh ok, I am sorry then, my bad, I never use the phone app so I never shared links from there. Sorry again @pooch-1

splendidmata, 2025-04-23:
I'd like to ask a question. How is a passive buzzer connected to the board? The + terminal is connected to 3.3V, the GND terminal is connected to T2, and where should the dummy pin be connected?

splendidmata, 2025-04-23:
谢谢您的回复，我的电子工程师朋友给了我一个建议，使用pdtd113zt替代扬蜂鸣器和振动电机的si2312，并取消t2与t3的10k电阻，有助于提高io口的寿命

pooch-1, 2025-04-23:
> > I believe that is only the shortened link. By default it looks like that when you want to share a link from the app.
> > […](#)
> 
> Oh ok, I am sorry then, my bad, I never use the phone app so I never shared links from there. Sorry again [@pooch-1](https://github.com/pooch-1)

No problem, appreciate the apology. Your work on this project is the reason I got interested so I appreciate your contributions. I am a novice but not a greedy one.

lupusworax, 2025-04-23:
> pdtd 113zt

how exactly is it increasing the life span of the Io port?   I am a bit confused how a resistor equipped transistor is going to do that?

meekowai, 2025-04-24:
Couldn’t get xb8089d0 so I got
https://a.aliexpress.com/_Ex6ODsq
https://a.aliexpress.com/_EyGML8I
Requires small change mentioned here https://github.com/gargomoma/fakeTec_pcb/issues/24#issuecomment-2796215177

splendidmata, 2025-04-25:
I have damaged several NiceNano this afternoon😭

mdbk95, 2025-04-27:
I get XB8789D0 with the same pinout but didnt work

zx-ecco, 2025-05-03:
> It is recommended to restore the direct plug welding form of niceNano. The yield rate of direct plug components for SMD is not very high.

not sure I understand what you mean by "direct plug welding"

lupusworax, 2025-05-03:
> > It is recommended to restore the direct plug welding form of niceNano. The yield rate of direct plug components for SMD is not very high.
> 
> not sure I understand what you mean by "direct plug welding"

Pretty sure he means THT, if you dont have a hot plate soldering the ProMicro on the V5 board by hand with a soldering iron.

JanitorHead, 2025-05-04:
The rev. B gerber file doesn't seem to work when uploaded to JLCPCB

ShimonHoranek, 2025-05-05:
@JanitorHead  You are right, uploaded a new archive. Thanks

JanitorHead, 2025-05-05:
> [@JanitorHead](https://github.com/JanitorHead) You are right, uploaded a new archive. Thanks

Nice, thank you!

[60 more replies]