gargomoma/fakeTec_pcb #25: RA-01SH

jgreusel1, 2025-03-25:
I have this device seemingly soldered in correctly. I have tried many of the firmwares and I always get Critical error #3 which is no LORA device. What am I doin wrong? Is the LORA device bricked?

Thanks,

John

jgreusel1, 2025-03-26:
I actually have a RA-01SH-P. Is that the problem?

deanZZZZZ, 2025-03-26:
I'm also thinking of getting the Ra-01SCH-P?
The VCCPA pin is different.
[link](https://www.houtenstaak.com/products/lora-ra-01sch-p.html)

jgreusel1, 2025-03-26:
Thought I had the -P version but the actual chip doesn't show that.  I'm beginning to think the Ra-01SH isn't really supported with the latest firmware.

deanZZZZZ, 2025-03-26:
Have you tried cutting the connection between TXEN and DIO2?

jgreusel1, 2025-03-26:
No, I ordered a few HT-RA62 instead.

jgreusel1, 2025-03-28:
This is resolved- bad trace on MISO pin- jumpered directly to the RA-01SH and it works fine!

iuvi7, 2025-03-28:
Can you upload gerbers for RA-01SH-H? Thanks

deanZZZZZ, 2025-04-18:
RA-01SH-P

RF_EN cut from RXEN
VCCPA cut from DIO2
VCCPA connect to 3v3
WORK with only battery connected
when connect usb and battery > not work
only usb > not work
connect VIN(3v3) to bench power supply > not work

not work means: does not send messages longer than 5 characters, everything else works without any errors

only on battery Range Test sends every 15 seconds with 100% success (29 dBm transmit power)

I don't understand why it doesn't work, if I directly connect the bench power supply to 3.3v VIN (nrf52)
Consumption when sending is about 700mA

iuvi7, 2025-06-19:
> RA-01SH-P
>
> RF_EN cut from RXEN VCCPA cut from DIO2 VCCPA connect to 3v3 WORK with only battery connected when connect usb and battery > not work only usb > not work connect VIN(3v3) to bench power [...]

Hi! Do you have any progress on today?

Can not find where MISO trace is bad than @jgreusel1 mentioned (what version of faketec?)

Thanks