gargomoma/fakeTec_pcb #26: fakeTec v6 - onboard solar charger

ShimonHoranek, 2025-04-09:
After a redesign of the back side of fakeTec v5 I present you fakeTec v6 with added CN3791 MPPT charger for solar panels. 
This version now features:

- CN3791 MPPT charger
- correct pads for GPS modules
- larger 6x3,6 mm buttons
- solar charger LEDs

I had to also do some compromises:

- only one mosfet for high current switching
- Battery can now be connected only through JST connector
- smaller 0805 components
There also wasn't space for adding dedicated encoder pads, will make v5 rev B.

Everything else is same as on v5 (if I didn't forget something)

Parts list:
 
- 2x 10M 0805 resistors
- 3x 10k 0805 resistors
- 1x 1k 0805 resistor
- 1x 30k 0805 resistor
- 5x 0,1uF 0805 capacitor
- 1x 4.7uF 0805 capacitor
- 1x 47uF 8x10 electrolytic capacitor
- 1x 10uF 106E tantalum capacitor

- 2x 6x3,6 SMD buttons
- 1x SI2312 mosfet
- 2x 1,25 JST connectors
- 1x CN3791 MPPT charger
- 1x green 0805 LED
- 1x red 0805 LED

- 2x SS54 schottky diodes
- 1x 100uH SDRH104R inductor
- 1x AO3401A mosfet

- 1x 2,2uF 0805 capacitor
- 1x 0,5R 4020 resistor
-1x 0806 Resistor based on your solar panel voltage:
6 V - 120k
9 V - 195k
12 V - 270k
18 V - 540k

violarulan, 2025-04-10:
> 1x 0,5R 4020 resistor

I couldn't find such package on aliexpress. Can you share the link to buy some? ...or is it ok to use 2512 (or some smaller package) and add some solder on both sides?

valzzu, 2025-04-14:
[image]
I saw this, u should fix this if u wanna use battery 😅
there's no way ur plugging a connector there.

NomDeTom, 2025-04-14:
Hey, @ShimonHoranek , this looks like an interesting design. I'm a huge fan of shoehorning stuff into small spaces on circuit boards, and this does a lot in a very small space.

Rather than run your design through a github issue, why not join us on the [Meshtastic discord](https://discord.com/channels/867578229534359593/1194757507013427250)? We can help you out with your design, and you can see the other stuff that people have worked on.

inakibur, 2025-05-15:
Hello @ShimonHoranek 
I am interested in this PCB to set up nodes with solar panels in my city.
Can gerber be created?
Is the PCB functional?
Greetings and thank you very much.

inakibur, 2025-05-16:
@valzzu For me the battery connector in that position is not a problem, if the connector cannot be connected I put the cable with tin on the plate.

valzzu, 2025-05-16:
it does look like it should work without any major issues but there prob is some ripple current and such things cuz there's not that many stitching vias.

inakibur, 2025-05-19:
One question, the resistance for the voltage of the solar panel is correct size 0805? Because the footprint on the PCB looks like 4020.
Greetings and thanks!

inakibur, 2025-05-21:
@ShimonHoranek Can't find to buy the XB8089D0, if I change the 1K resistance to a 100OHM resistance, can I use the XB8789D0?
Greetings and thanks!

lupusworax, 2025-05-22:
> [@ShimonHoranek](https://github.com/ShimonHoranek) Can't find to buy the XB8089D0, if I change the 1K resistance to a 100OHM resistance, can I use the XB8789D0? Greetings and thanks!

No clue if the listing is available in your country, often having problems sharing links to ali  but anyways does this seller shipt to your country? 
https://de.aliexpress.com/item/1005005993725347.html

inakibur, 2025-05-22:
> No clue if the listing is available in your country, often having problems sharing links to ali but anyways does this seller shipt to your country? https://de.aliexpress.com/item/1005005993725347.html

Hello
This seller does send to my country, but I only get the XB8789D0 and in the datasheet he puts that it carries a resistance of 100 ohm in the place of 1k ohm.
Greetings and thanks!

inakibur, 2025-06-04:
Buenas noches
Hoy he montado la faketec v6. La placa enciende bien y funciona la bateria.
Para probar la carga solar he conectado una fuente de alimentación de 6v 500mha pero no carga la bateria, se enciende el Led rojo pero no hay carga en la medición de corriente en la bateria.
No soy capaz de encontrar el fallo.
Saludos y gracias

[image]
[image]

inakibur, 2025-06-04:
I have measured the CN3791 test values and they seem correct, tomorrow I will continue to test.
Greetings!

inakibur, 2025-06-05:
In the two pads marked with an arrow there is continuity with the polymeter and I think there should not be one, am I correct?
Greetings and thanks!

[image]

violarulan, 2025-06-06:
> Update: Following the clues, I managed to correct the failure of the XB8789D0 by piercing the dot marked with a red circle in the photo.

What do you mean by piercing the dot? Do I have to do the same if I use XB8089D0?

Thanks.

lupusworax, 2025-06-06:
> > Update: Following the clues, I managed to correct the failure of the XB8789D0 by piercing the dot marked with a red circle in the photo.
> 
> What do you mean by piercing the dot? Do I have to do the same if I use XB8089D0?
> 
> Thanks.

Is this about fixing, not switching off ones reaching 2.9volt cutoff voltage?

NomDeTom, 2025-06-09:
It might be worth reviewing the choices of the bias resistors for the mosfets and the battery voltage divider.
If all three mosfets are on, they will be drawing a total of 1mA (0.33mA each), which is quite high compared to the total idle consumption of a promicro node (~10mA).

For the battery voltage divider, the default in the firmware is 3:2, and I've generally found that ~1M is enough - 10M is certainly very high. You should consider using either 1.5M and 1M, or even lower, like 500k (510k would be fine) and 750k.

Finally, the user button and reset do not need pull-up resistors - the NRF is capable of applying its own internally. The 100nF decoupling capacitors should be placed as close to the buttons as possible - consider moving them to the top side, even if they have to be hand assembled with the switches.

NomDeTom, 2025-06-09:
Also, the resistor for the MPPT selector for 18V should be 420k, not 540k (assuming you're using a 30k resistor).

valzzu, 2025-06-09:
> In the two pads marked with an arrow there is continuity with the polymeter and I think there should not be one, am I correct? Greetings and thanks!
> 
> [image]
> 
> Update: Following the clues, I managed to correct the failure of the XB8789D0 by piercing the dot marked with a red circle in the photo.
> 
> [image] Tomorrow I will upload a video with the faketec v6 working and all the verifications made. Greetings and thanks.

coming back to this, the ic is wired wrong.
take a look at my schema
https://zipline.valzzu.xyz/u/4Tg2iG.png
vs
https://zipline.valzzu.xyz/u/ixduWL.png

valzzu, 2025-06-09:
nvm, not sure, it looks wrong but now that i think of it not completly sure

valzzu, 2025-06-09:
nvm ye this is the real issue, via in the wrong place
https://zipline.valzzu.xyz/u/mwecjO.png

inakibur, 2025-06-09:
> this is the real issue, via in the wrong place https://zipline.valzzu.xyz/u/mwecjO.png

Si, ese es el punto que he tenido que quitar con una broca y el taladro, despues de eso funciona todo bien en la V6.
Saludos y muchas gracias.

ShimonHoranek, 2025-07-04:
Hey, sorry for not being active and thanks for testing this board. Do I understand it right that the current issues are:

- wrongly placed via on the XB
- wrong position of JST
- wrong size of the 0,5 Ohm resistor package

If you can send me the dimensions of the resistor used in the typical solar charger boards, I can definitely fix all three of them.

I'm using 10 MOhm resistors for battery measurement to minimize the current and thus power consumption. I have added a parallel cap that should charge to the halved battery voltage and thus you shouldn't need higher current. But replacing them with smaller is not hard. If you found out that smaller resistances work much better I will update the label and the post.

I have included the pull up resistor for the buttons as suggested by shurillu in a V5 comment. He included schematics of other boards using nRF52 and they both used them.

The decoupling caps can be probably moved a bit closer.

I'm happy that the board works (be it with the BMS via fixed). It was mostly done as an experiment and finding that @lupusworax used V5 with a solar panel directly I figured this board would not be useful. But if there are no other problems I will look into improving the design.

BastianWagner, 2025-07-26:
What are the minimal components you need, if I would power the board via USB only? So no battery or solar, but would still like to use the v6 board in case I would like to upgrade to solar+battery.

iuvi7, 2025-07-27:
Fixed version will be released as updated here, or as new one as v6.01? Thanks

meekowai, 2025-08-01:
I think there is a mistake in description layer - 100uH points to electrolitic cap instead of coil. Am I right?

BastianWagner, 2025-08-04:
Would be interested too!

NomDeTom, 2025-08-05:
> Hello, I found that R1 and R2 are not too high to choose 10m, connecting the battery and solar panel cannot operate normally, they can be operated by replacing them with 2m, but nrf52 will overheat, so that parameter is more suitable.

@lunneswang do you mean 10m and 2m as in 10 milliohms or megaohms? They should be above 100kohms ideally, but above 1Mohm (i.e. 1 000 000 ohms) it is diminishing returns.

lunneswang, 2025-08-05:
> > Hello, I found that R1 and R2 are not too high to choose 10m, connecting the battery and solar panel cannot operate normally, they can be operated by replacing them with 2m, but nrf52 will overheat, so that parameter is more suitable.您好，我发现 R1 和 R2 选择 10m 都不太高，连接电池和太阳能电池板不能正常运行，换成 2m 就可以运行，但是 nrf52 会过热，这样参数比较合适。
> 
> [@lunneswang](https://github.com/lunneswang) do you mean 10m and 2m as in 10 milliohms or megaohms? They should be above 100kohms ideally, but above 1Mohm (i.e. 1 000 000 ohms) it is diminishing returns.您指的是 10m 和 2m，如 10 毫欧还是兆欧？理想情况下，它们应该高于 100 毫欧姆，但高于 1Mohm（即 1 000 000 欧姆）就会递减回报。
According to the original schematic, it is 10 megohms. I have tried using 1M and 2M, and both work. However, the nrf52 will get very hot when plugged into the typec, so I am not sure which one to choose or how to measure to get the result.

SrLPZ, 2025-09-06:
The tantalum capacitor polarity is reversed, be careful. 

Otherwise, everything's fine. Thank you!

Nombrealeatorio, 2025-09-27:
> The tantalum capacitor polarity is reversed, be careful.
> 
> Otherwise, everything's fine. Thank you!

Do you mean this is the correct way? Can I ask if you've tested the fakeTec_V6_revB ?

[image]

ngarafol, 2025-09-28:
> > The tantalum capacitor polarity is reversed, be careful.
> > Otherwise, everything's fine. Thank you!
> 
> Do you mean this is the correct way? Can I ask if you've tested the fakeTec_V6_revB ?
> 
> <img alt="Image" width="570" height="645" src="https://private-user-images.githubusercontent.com/128269626/494723187-9ea27464-7ce2-4437-8714-5fb5ad67a6b2.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTkwNjU4MDksIm5iZiI6MTc1OTA2NTUwOSwicGF0aCI6Ii8xMjgyNjk2MjYvNDk0NzIzMTg3LTllYTI3NDY0LTdjZTItNDQzNy04NzE0LTVmYjVhZDY3YTZiMi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUwOTI4JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MDkyOFQxMzE4MjlaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1jZTZkOTdkY2ZkODA1NjM3NjJmMzExNWFiZGYxNTM2Y2NhOGQ1YmNjY2YyMWM5NmU5OWIzZDQzYjQ5NzgwYTk3JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.HMDfxnG0QjnHY1ide0Qh8jsqKxnyIuop3gWFD7LgxHA">

Indeed, good question. In pictures above from inakibur, it is mounted as marked on PCB (+ on the right), so perhaps SrLPZ warned about 3d viewer in Kicad having wrong orientation for tantalum capacitor of 10uF 🤔

SrLPZ, 2025-09-28:
> Do you mean this is the correct way? Can I ask if you've tested the fakeTec_V6_revB ?
>
> <img alt="Image" width="570" height="645" [...]

Yes, I'm using fakeTec_V6_revB. The positive is on the left side as you have indicated in the image.

> Indeed, good question. In pictures above from inakibur, it is mounted as marked on PCB (+ on the right), so perhaps SrLPZ warned about 3d viewer in Kicad having wrong orientation for tantalum [...]

This is a small typo error in Rev. B only, in the Rev. A the marking is correct.

ShimonHoranek, 2025-09-28:
@SrLPZ thanks for noticing! It is indeed marked incorrectly. I have now fixed that and uploaded rev C. Have you tested rev B and can confirm everything works fine? I still have not ordered any myself so I can't test them :D 

@ngarafol inakibur is using rev. A boards in those pictures as rev. B didn't exist at that time. The capacitor is in the same place on both versions, but it is flipped. You can see that in the pcb files in kicad. I don't remember why I flipped it, but it sure had some reason. Electrically it is wired up still the same

Toshik, 2025-12-03:
Hey @ShimonHoranek , thank you for the board design.
I've noticed C9 shows 2.2uF on the silk and 220nF in the circuit, which one supposed to be correct one?

iuvi7, 2025-12-18:
@shimonomihs wouldn't it be cool to add one capacitor to ht-ra62 module give it some buffer for power delivery and also filtering? Also on v5 as well

[image]

gretel, 2026-01-04:
during 39C3 some boards were built, working fine!
only change is using `XB8089D0` with 1k resistor.

[fakeTec_V6c_BOM.pdf](https://github.com/user-attachments/files/24419471/fakeTec_V6c_BOM.pdf)
[fakeTec_V6c.pdf](https://github.com/user-attachments/files/24419490/fakeTec_V6c.pdf)
[image]

treysis, 2026-01-29:
> during 39C3 some boards were built, working fine!

Still haven't completed mine 🙄 BUT: regarding the display, there seem to be some display modules that have reversed GND/VCC, i.e. VCC/GND/SCL/SDA. That includes the ideaspark boards (round case) you handed out at 39C3. All my other displays or in general I2C modules I have at home have the same pin out as the faketec v6 rev. C though.

treysis, 2026-01-29:
@gretel no, much easier to just order a fitting display. Just wanted to let you know that you might have a couple of displays in your stash that are not compatible with the faketec design.

treysis, 2026-01-31:
Ah, I might have overheard this 😀 yeah, with cables it's not problem of course. Ghen disregard my comment. Thx for the workshop though, it was one of the best in my opinion.

SET19724, 2026-02-10:
> Could you please tell me if this version includes GPS and vibration? And what contacts (pins) are these?
BOARD>GPS
T1>GND
TX>TX
RX>RX
3.3V>VCC

Vibration i not tested yet

ogzer707, 2026-03-28:
[image]

Does anybody see anything off here? 
My battery positive has continuity with polarized parts ground instead of the positive. My FakeTec v6 rev c only works through usb power. Not battery. This is the second one I’ve made and did a cleaner job this time and triple checked polarity and resistance to all parts. But this one is messing with me. Any help would be great👍🏽

ogzer707, 2026-03-28:
Ok. Thank you. I’ll be back home in 20 to give that a try. And yes I bought jst connectors from Amazon and they aren’t what I wanted. But I’m making it work for now. Thanks for your time I’ll post back what I find…. 
Oh I also took jst connector off and tried ohming out from pads and same results so I don’t think it’s the connector itself.

nnamhcor, 2026-09-09:
As an experiment, I added a 10-ohm resistor between pin 10 (DRV) of the CN3791 chip. Аfter lifting the leg off the PCB and the gate of the AO3401A. It seems this reduces the level of parasitic noise. And I realize this is a deviation from the datasheet.

After that, my node ran for 11 full days. For reference, before that, it couldn't even run for a full day without glitches.

[image]
[image]

[image]
[image]

This is probably not the ideal solution to the problem that I and other users in this thread have encountered.

I suggest adding pins for the new 10-ohm resistor. If everything works fine for future users without it, these pins can simply be shorted together (similar to the choice between BMS and Direct modes).

I’m happy to invite everyone to continue testing and discussing this idea!