gargomoma/fakeTec_pcb #27: Does anyone have ESP32 firmware to save ProMicro nRF52840??

JohnTaipei, 2025-04-15:
As title
Thanks :-)

shurillu, 2025-04-16:
Hello JohnTaipei, 

   I don't know exactly what you want to do, but the ESP32 and the nRF52840 are different platforms so a firmware that run in one does not run in the other.

Stefano

mkw432, 2025-04-16:
I think @JohnTaipei is asking if anyone has built the firmware here that they could share: https://github.com/atc1441/ESP32_nRF52_SWD

@JohnTaipei - the ESP32 flasher firmware may need to be tailored to the specific ESP32 board you're using. Send me a message on discord (same username) with a link to your ESP32 and I'll try to help you get what you need.

gargomoma, 2025-04-16:
Hi @JohnTaipei i have the fw built for esp32.
I'll look for it and upload it to the repo
Cheers,

gargomoma, 2025-04-17:
esptool.py --chip esp32 --port XXXX write_flash -z 0x0 ESP32_nRF52_SWD-ESP32_firmware.bin
Wifi pass: "c3d42c457715da2a" (Without the quotation marks)

[ESP32_nRF52_SWD-ESP32_firmware_bin.txt](https://github.com/user-attachments/files/19786185/ESP32_nRF52_SWD-ESP32_firmware_bin.txt)

Fix the filename. remove .txt to .bin

JohnTaipei, 2025-04-17:
Thank you so much
@gargomoma 
@mkw432