gargomoma/fakeTec_pcb #29: Currently working on adapting the AMADE2 portable node  to the Faketec boards but need some help with remaining free GPIO pins.

lupusworax, 2025-05-05:
So fare everything went straight forward but now I am missing one last option, the LED light bar needs one more free GPIO pin and at this point i am not even sure if the Pro Micro NRF board does have any pins free?  I use the board with button, vibra motor, buzzer, GPS enable and rotary encoder, so using pin 0,6, 8, 24, 33, 34, 39. 
I need one last pin for the notification led and it would be perfect but I think there are no more free pins/ connections? Can anyone help me out here if  there is still some available spot free for one last addon?

[image]
[image]

timurey, 2025-05-05:
How about build-in led? P0.15?

lupusworax, 2025-05-05:
> How about build-in led? P0.15?

hmm I would prefer a free pin for the big red LED bar and dont wont it to show any heartbeat or "status" action. So is there even a free pin left over that is accessible or are they all used up with with my configuration?

lupusworax, 2025-05-05:
could it be that 

[image]
is unused? I tried to test continuity from the solder pad  to all other pads and could not find any?

lupusworax, 2025-05-05:
haha yes the very last free gpio pin and it works :D

lupusworax, 2025-05-05:
> Are the pins 101, 102 and 107 defined as 33, 34 and 39 in meshtastic configuration?

correct, 1.0X +32 =config number