gargomoma/fakeTec_pcb #30: Failrate on ProMicro NRF52 dev boards from TenstarRobot on ali express ( the bright red ones)

lupusworax, 2025-05-08:
The well known double packs from TenstarRobot on ali express with their bright red PCB is the cheapest option to buy a ProMicro board on ali.  A double pack goes for 5-6 euros. 
I got myself a pack of 20 (40 boards), first I thought I am unlucky or doing something wrong but from 20-25 board I already built to finished faketec boards 10! boards started to fail short after or right after soldering. 

I am using a low temp solder when using the hotplate with a melting point of 138°Celsisus, far from a temperature that would damage components.  
I am also not cooking it for too long, I use loads of flux and there are 100% no shorts but many of them keep failing in the process. 

I of course also change the RA-62 chip to check if maybe it is the culprit but so far not a single one of them failed, Heltec knows how to build stuff!! 
Even after getting the Pro micro board off the heltec board to check if it works away from it and not soldered to anything with all the contacts perfectly fine and nothing bridged, they just stopped working.  

5 boards ago I started to switch to an older badge of boards ( single packed black pcb) I got from a different vendor on ali. 
No changes in the way I build them ( by now I have made 50 + boards)  none of them failed. 

So I wanted to ask other users as I think many people go for the red double pack from TenstarRobot. 
Anyone else having a high fail rate? Or am I just unlucky? 

Or is there a reason it comes so cheap in a back of two?!

zoranignjatovic, 2025-05-09:
can you link that other vendor?

lupusworax, 2025-05-09:
https://de.aliexpress.com/item/1005007383270623.html

zx-ecco, 2025-05-10:
any chance it could be from the pcb itself? like something in the component placement that doesn't agree with the red ones?

violarulan, 2025-05-11:
As heard from others, some cheap promicros are 'reused' ones. 

I bought 15 promicros from like 4-5 resellers at ~2 usd for each _(Yes, they are cheaper here in China)_. Two of them from two different resellers are dead. If you power it on, it keeps reseting itself (red led is blinking fast). STLink can flash the bootloader without any issue. But no matter how many times you double-short the RST, it's still dead.

In taobao(China mainland version of AE), 100% geniune well-functioning promicros come from one reseller at ~3.5 USD, which is a little more expensive.

**Again, test your promicros before soldering!**

JohnTaipei, 2025-05-18:
buy 5 pcs
failes 2 pcs

NoHaxJustParker, 2025-05-20:
I also have had 1 out of the 2 red pro-micros that I have fail (It seems to use the same charge controller as the heltec v3 which has a problem with being charged with some Samsung fast charging bricks (a Samsung fast charging brick caused my heltec v3 to die))

ngavars, 2025-06-03:
I've had two of the red Promicro boards fail on me right after soldering. They were fine before soldering, because I always check them and also update the boot loader. I blamed it on static discharge, as I solder by hand and did not use a wrist strap. But who knows. Quality wise, I have seen two boards where the tiny 0402 resistors (or caps, don't remember) were soldered vertically - as in one end soldered to a pad and the other end is pointing upwards.
Anyhow - from 50+ boards that I have put together, only two duds is a good run.

wilqp, 2025-07-08:
I also have another problem with those boards- I can't make them work with the Li-Po 3.7v batteries. It doesn't matter if connecting directly, or via fakeTec v5 (direct, or BMS).

lupusworax, 2025-07-08:
> I also have another problem with those boards- I can't make them work with the Li-Po 3.7v batteries. It doesn't matter if connecting directly, or via fakeTec v5 (direct, or BMS).

can you provide pictures of your setup?

wilqp, 2025-07-10:
[image]

[image]

[image]

Hi,
Here they are, I think that I tried every combination,
With the NiceNano V2 out of the box(factory firmware) I tried using following pins RAW&GND, B+&B-, jst1.25. On FakeTec v5 board I tried combining it with 1k R, changing it to 100 R (I'm usingXB8089D- without"0"), also the with pads for BMS/direct(always just one pad connected with a middle pad).
It's the same with like those on the FakeTec V5 pcb - they work only on power via USB-C, and with/without OLED.
I assume it doesn't have power/charging chip?

On the side-note, I bought 14 more, and just 1 was faulty(I couldn't update& install meshtastic software).

lupusworax, 2025-07-10:
yeah I think I see the problem, I always had to resolder the pro micros by hand,  go over all the connection points with a soldering iron  and flux, and dont go easy on the heat + solder, make sure the connections are properly going through all the way. I also thought it works all great with the hotplate  but in the end it only started working after going each of the spots and add a proper amount of solder as I had some cold spots.

[image]

lupusworax, 2025-07-12:
> TL;DR: Occam's razor- not all versions have the charging/power chip.
> 
> After soldering, I use multimeter & 1 needle to check the connectivity of the fakeTec PCB's pads with the soldered board top. After checking with MPPT, Li-Po charging modules, DC-DC step-up’s, and in different pins, schematics, and messaging that was another sub-version. One with without power/charging chip. Their USBs’ have very little voltage tolerance- they need AT LEAST 5V (5~5.1V?), as the voltage of 4.9 is not enough to work. It might be a good idea to add in repo info that there’s another subversion without charging. Ex: https://aliexpress.com/item/1005007738886550.html

Without charging? I never had one that did not charge. I can see on one of your pictures that you did not soldered the charging boost pads so its only charging with 1/3 of what it is capable off. 

[image]

gretel, 2025-11-12:
> I reflow dead red ProMicro NRF52 at 270C (I moved the nrf52 a little with tweezers) and now it is alive and working. Before that I was able to upload bootloader successfully with st-link-v2 but promicro was not working.
> <img alt="Image" width="594" height="741" src="https://private-user-images.githubusercontent.com/95451596/502875111-8c9f4938-7abd-444b-b0c2-113a8b829f21.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NjI5ODgxNDgsIm5iZiI6MTc2Mjk4Nzg0OCwicGF0aCI6Ii85NTQ1MTU5Ni81MDI4NzUxMTEtOGM5ZjQ5MzgtN2FiZC00NDRiLWIwYzItMTEzYThiODI5ZjIxLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTExMTIlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUxMTEyVDIyNTA0OFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWZmMGNiYTY5MjdhZmIxY2RjMmZkMGNkNzk1ZjU0ZjAyYjg1ZjYyYmEwM2ExMTEzZDE2N2E5MjI0N2EyZjdlMDcmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.4YjKxoFlLcjqsWs7t_CysDsRkBpr8LwdWuSSX6W0vz4">

according to my [favourite source](https://github.com/joric/nrfmicro/wiki/Alternatives#supermini-nrf52840-red) the blacks ones have been failing previously and the red ones should be fine.. thanks for the sharing.