gargomoma/fakeTec_pcb #31: I2C devices not detected on Faketec V3 (SSD1306 / BME280)

juanpintom, 2025-05-19:
Hi,

I'm trying to connect an SSD1306 OLED display via I2C to my Faketec V3 board, but no I2C devices are detected. I've tested with multiple displays and also with a BME280 sensor, and in all cases, the result is the same: no I2C devices found.

Here's the relevant part of the serial log:

2025-05-19 21:08:17.200: INFO  | ??:??:?? 1 Scan for i2c devices
2025-05-19 21:08:17.203: DEBUG | ??:??:?? 1 Scan for I2C devices on port 1
2025-05-19 21:08:17.205: INFO  | ??:??:?? 1 No I2C devices found

What I've tried:

Different SSD1306 displays (both 128x32 and 128x64)

A known working BME280 sensor

Verified wiring and pin continuity with a multimeter

Double-checked the SDA and SCL pins according to the board's documentation

None of the devices are detected during the I2C scan. Is there any known issue with I2C on the Faketec V3 or any configuration I might be missing?

Thanks in advance!

juanpintom, 2025-05-20:
Apologies for the noise — I’ve just tested the exact same setup (same SSD1306 and BME280) with a different PCB and a different nRF52 module, and everything works as expected. 

Thanks again!

lupusworax, 2025-05-20:
No worries, just confirms that those Pro micro boards dont use the best quality parts ;)  Also I never had a RA-62 module fail on me, so keep a hold on that and test it with another pro micro as it probably works just fine ;)