gargomoma/fakeTec_pcb #37: Issue with powerful RA-01S-P (29dB)

iuvi7, 2025-06-24:
Hey everyone ! 

Running for a compact more powerful use of pocket-node, I have an issue with using RA-01S-P (29dB) module, so i decided to ask if anyone else had tested and have success with them on faketec or esp32?

Problem is - it's working, but sending longer messages failing

After fails, i try to split power. Give 3.3v to a module, and separately to NRF but resul was same

By datasheet it said "5dB" setting should be set up, otherwise it can be burned 

Thanks  

[image]

sbeardPA, 2025-07-25:
This appears to be an issue from the lack of TXCO in this chip coupled with the high output. Th RA-01SH also lacks a txco but is lower power so less temp. Long _fast is 3 times the bit rate of Long_moderate so the transmit requires more power. If you put a board in the freezer for 30 minutes you can transmit long messages successfully. Just confirmed it with 2 Faketec V4 boards

LiveJay, 2026-04-03:
It sounds like this is a hardware problem.  Do we think there is going to be a fix or should we abandon this device?