gargomoma/fakeTec_pcb #40: New Faketec based node design "FakeTec Case - Testosterone Edition"

lastparody, 2025-07-28:
Hi everyone!
I’ve just finished designing a 3D-printable case for the FakeTec PCB and wanted to share it with the community.

🧩 The model includes support for:

1.3" OLED display
GPS module
BME280 sensor
On/off slide switch
103450 LiPo battery (3.7V 2000 mAh)

There are three different versions of the case:

Full version (OLED + BME280 + switch)
No display/switch, with BME280
Plain version (just the PCB housing)

🔗 [View & Download on MakerWorld]( https://makerworld.com/tr/models/1646395-faketec-case-testosterone-edition)

I designed this with a little fun twist – it kind of looks like a small gas can 😄
I'm happy to share it and would love to hear your feedback or suggestions. Hope it helps others working on FakeTec-based builds!

[image]

lastparody, 2025-08-01:
Hello, I wonder if you can add my project to the [Community Contributions](https://github.com/gargomoma/fakeTec_pcb?tab=readme-ov-file#community-contributions) section, thank you.  👉👈

@porkcube @elmanytas @Kongduino

lastparody, 2025-08-01:
Hello, I wonder if you can add my project to the [Community Contributions](https://github.com/gargomoma/fakeTec_pcb?tab=readme-ov-file#community-contributions) section, thank you.  👉👈

@gargomoma @porkcube @elmanytas @Kongduino

lupusworax, 2025-08-28:
Very cool design, love it, I just printed it, building it as I write, sadly I only have a 1.3 sideway display at hand that is not yet compatible with meshtastic. but everything else fits perfectly together, great work!